sparkfun/SparkFun_AS3935_Lightning_Detector_Arduino_Library #1: malformed if() clauses

wjcarpenter, 2019-04-13:
In the library code, there are some instances of IF statements at the beginnings of methods to check for legal values. For example, checking if parameter "a" is one of the legal values 1 or 2:

**_if ((a!=1) | ( a!=2)) return;_**

Those IF statements can never be true because of the bitwise OR operator. As a result, those methods always silently return without doing anything.

wjcarpenter, 2019-04-18:
Sorry, I meant "are always TRUE", hence the return is always taken.

nseidle, 2019-04-18:
Nice find! Thank you. The engineer behind this repo is on paternity leave. We'll get this cleaned up asap. If you or anyone else wants to put a PR together we would love to review it.

For my own notes, lines effected:

https://github.com/sparkfun/SparkFun_AS3935_Lightning_Detector_Library/blob/master/src/SparkFun_AS3935.cpp#L85

https://github.com/sparkfun/SparkFun_AS3935_Lightning_Detector_Library/blob/master/src/SparkFun_AS3935.cpp#L139

https://github.com/sparkfun/SparkFun_AS3935_Lightning_Detector_Library/blob/master/src/SparkFun_AS3935.cpp#L203

Let me know if you see any others.

edspark, 2019-04-22:
Thanks for the issue @wjcarpenter  and thanks for highlighting the specific lines @nseidle. Those particular lines have been fixed in this commit: 
https://github.com/sparkfun/SparkFun_AS3935_Lightning_Detector_Library/commit/d33f3f3cfb031a4116759189dd24c766d15bff94

closed 2019-04-22 by edspark