sparkfun/SparkFun_AS3935_Lightning_Detector_Arduino_Library #7: Esp32 compiling error

takenalias, 2019-06-19:
Hi, I am running version 1.3.3 of this library, but it is still not compiling and is coming up with the following error. 

I am having this issue on an esp32, no modifications to the example script except the interrupt pin number. 

Arduino: 1.8.9 (Windows 10), Board: "Adafruit ESP32 Feather, 80MHz, 921600, None"

Example2_More_Lightning_Features_I2C:33:38: error: invalid conversion from 'int' to 'SF_AS3935_I2C_ADDRESS' [-fpermissive]

 SparkFun_AS3935 lightning(AS3935_ADDR);

                                      ^

In file included from C:\Users\x\Documents\Arduino\libraries\SparkFun_AS3935_Lightning_Detector_Arduino_Library\examples\Example2_More_Lightning_Features_I2C\Example2_More_Lightning_Features_I2C.ino:19:0:

C:\Users\x\Documents\Arduino\libraries\SparkFun_AS3935_Lightning_Detector_Arduino_Library\src/SparkFun_AS3935.h:77:5: note:   initializing argument 1 of 'SparkFun_AS3935::SparkFun_AS3935(SF_AS3935_I2C_ADDRESS)'

     SparkFun_AS3935(enum SF_AS3935_I2C_ADDRESS address);

     ^

exit status 1
invalid conversion from 'int' to 'SF_AS3935_I2C_ADDRESS' [-fpermissive]

gon0, 2019-06-20:
Same on ESP8266 LOLIN D1 mini pro:

```
Example1_BasicLightning_I2C:30:38: error: invalid conversion from 'int' to 'SF_AS3935_I2C_ADDRESS' [-fpermissive]

 SparkFun_AS3935 lightning(AS3935_ADDR);

                                      ^

In file included from C:\Users\x\AppData\Local\Temp\arduino_modified_sketch_616929\Example1_BasicLightning_I2C.ino:18:0:

C:\Users\x\Documents\Arduino\libraries\SparkFun_AS3935_Lightning_Detector_Arduino_Library\src/SparkFun_AS3935.h:77:5: error:   initializing argument 1 of 'SparkFun_AS3935::SparkFun_AS3935(SF_AS3935_I2C_ADDRESS)' [-fpermissive]

     SparkFun_AS3935(enum SF_AS3935_I2C_ADDRESS address);

     ^

exit status 1
invalid conversion from 'int' to 'SF_AS3935_I2C_ADDRESS' [-fpermissive]
```

edspark, 2019-06-20:
Thank you the issue @takenalias and for giving a work around @gon0. I have fixed the issue with the latest release v1.3.4, so a work around is no long necessary.

closed 2019-06-20 by edspark