sparkfun/SparkFun_AS3935_Lightning_Detector_Arduino_Library #10: Wake up does not write to register 0x08 bit 5 1, wait 2ms them write 0 after calibration

fourstix, 2019-07-09:
Hi,
1) I noticed that the wakeUp() routine does not write to register 0x09 bit 5 after the Calibration command, like the datasheet on page 23, Section 8.11 says it should.

Here's the text from the datasheet.

> If the AS3935 is set in power-down mode, the TRCO needs to be recalibrated using the following procedure:
> 1. Send Direct command CALIB_RCO
> 2. Modify REG0x08[5] = 1
> 3. Wait 2ms
> 4. Modify REG0x08[5] = 0

I added the reg 0x08 bit 5 write  1, wait 2ms, write 0 logic to my python port of your Arduino code and it seems to work fine.  I also compared with similar code in the pcfens/RPi_AS3539 python project calibration routine and I noticed that code has the bit 5, write 1, wait 2ms, write 0 logic.  I have tested my python code with both SPI and I2C.  I wonder if this should be added to the Arduino code as well, so I opened this issue to let you know.

I also have a couple more questions.

2) I can't find CALIB_SRCO register 0x3B documented anywhere in the datasheet.  However I do get a value that indicates true after calibration, so I went ahead and kept the check in the ported code.  It seems to work fine, and I'm guessing you have documentation, and that's why the code checks this byte.  Is that correct?

3) The pcfens/RPi_AS3539 code has a calibrate() and reset() routines.  That seemed like a good idea to make these available for tuning, so I added these to the ported python code.  You might want to add them to the Arduino code, as they are trivial and reuse code you already have.

If you'd like to  take a look a the CircuitPython port, it's available at
[fourstix/Sparkfun_CircuitPython_QwiicAS3935](https://github.com/fourstix/Sparkfun_CircuitPython_QwiicAS3935)
Your comments and suggestions are welcome.  I hope to release it to the CircuitPython community bundle soon.

edspark, 2019-07-10:
I'll modify the library/example code soon - early next week. 
1. After double checking the datasheet it should indeed be as you stated/linked. 
2. Make sure you have at least v1.04 of the AS3935 datasheet. Here on page 20 is the registers in question:
[image]

On page 36 they specify checking this particular register to check if calibration for both oscillators is done and if it was successful.

3. Sounds like a great idea!

I'll close this out when this gets updated. Thanks @fourstix!

fourstix, 2019-07-14:
Ah, thank you, that's it.  The datasheet I have from the link under the Documents section on the Sparkfun's product page is 1.02.

edspark, 2019-07-17:
The wakeup function was fixed and the functions you mentioned were added in this commit: https://github.com/sparkfun/SparkFun_AS3935_Lightning_Detector_Arduino_Library/commit/561c46db3a96cc48aa65045f9e85184a34273c33, as well as a host of other changes. I'll get the latest datasheet added to our SparkFun page shortly.  Thanks for your CircuitPython contribution and for your contribution to this library!

closed 2019-07-17 by edspark