sparkfun/SparkFun_AS3935_Lightning_Detector_Arduino_Library #12: Missing license file

buelowp, 2019-08-11:
I would like to use or if needed, provide a Particle library for this chip using SPI. Right now, you are missing a license file, and I can't exactly glean which license I could use which means I can't create a Particle library. Can you please update the repository with a correct license? Alternatively, can you please provide an official searchable Particle library for use with their build tools? Thank you.

edspark, 2019-08-12:
Ack! MIT license has been added for this library. Sorry about the confusion there.

closed 2019-08-12 by edspark

buelowp, 2019-08-12:
Thank you. Do you care or mind that I have uploaded a library to the Particle build system for this? I tried to name it such that you could do it as well without conflict. Mine is currently private, but i do plan to make it public in the near future. I would obsolete mine in the event you choose to support Particle in this way. The reason is that you have the only SPI version available for Particle and it works very well on that chip. Thank you very much.

edspark, 2019-08-12:
Yes of course! Just be sure to attribute the original source of the code when possible, i.e. SparkFun Electronics. Happy hacking!