sparkfun/SparkFun_AS3935_Lightning_Detector_Arduino_Library #16: How to modify this library to work with particle argon (or photon)

mrsoltys, 2020-08-11:
### Subject of the issue
Library doesn't compile on Particle Argon

### Expected behavior
If I include the .cpp and .h files and try to compile on an argon, I would like for it to compile so I can use this library

### Actual behavior
Throws a tonne of errors, starting with:
n file included from ../platform/MCU/nRF52840/inc/hw_config.h:11:0,
                 from ../hal/inc/usb_hal.h:43,
                 from ../wiring/inc/spark_wiring_usbserial.h:32,
                 from ./inc/application.h:51,
                 from ./inc/Particle.h:5,
                 from ./inc/Arduino.h:11,
                 from ./inc/Wire.h:1,
                 from lib/SparkFun_AS3935_Lightning_Detector_Arduino_Library/src/SparkFun_AS3935_Lightning_Detector_Arduino_Library.h:4,
                 from lib/SparkFun_AS3935_Lightning_Detector_Arduino_Library/src/SparkFun_AS3935_Lightning_Detector_Arduino_Library.cpp:12:
../platform/MCU/nRF52840/inc/hw_system_flags.h:32:33: error: expected identifier before numeric constant
 #define RESET                   0
                                 ^
lib/SparkFun_AS3935_Lightning_Detector_Arduino_Library/src/SparkFun_AS3935_Lightning_Detector_Arduino_Library.h:27:3: note: in expansion of macro 'RESET'
   RESET             = 0x3C,
   ^
../platform/MCU/nRF52840/inc/hw_system_flags.h:32:33: error: expected '}' before numeric constant
 #define RESET                   0

edspark, 2020-09-02:
Hi! I think this issue is occurring because the constant `RESET` from this library is conflicting with a constant of the same name within **hw_system_flags.h**. I've changed `RESET` to `RESET_LIGHT` in this library, it's probably too generic anyway, to hopefully fix the issue you're seeing. Open this back up if that doesn't work.

edspark, 2020-09-02:
Fixed in the latest [release](https://github.com/sparkfun/SparkFun_AS3935_Lightning_Detector_Arduino_Library/releases/tag/v1.4.2)

closed 2020-09-02 by edspark