sparkfun/SparkFun_AS3935_Lightning_Detector_Arduino_Library #17: Disturbers not being masked

ojones117, 2020-10-18:
### Subject of the issue
Getting lots of disturbers on the serial output. Even with mask disturbers being true. Using code revision 1e1a717
Haven't detected any lightning events even with a storm passing by. Wondering if something isn't configured correctly.

### Your workbench
* What development board or microcontroller are you using?
Arduino Nano
* What version of hardware or breakout board are you using? 
Sparkfun AS3935 SEN-15441
* How is the breakout board wired to your microcontroller?
SPI
* How is everything being powered?
PC USB
* Are there any additional details that may help us help you?

### Steps to reproduce
Tell us how to reproduce this issue. Please post stripped down example code demonstrating your issue.
Using "Example2_More_Lightning_Features_SPI

lightning.maskDisturber(true); 

### Expected behavior
Not see Disturber events on the serial output

### Actual behavior

AS3935 Franklin Lightning Detector
Schmow-ZoW, Lightning Detector Ready!

Are disturbers being masked: YES
Are we set for indoor or outdoor: Indoor.
Noise Level is set at: 3
Watchdog Threshold is set to: 2
Spike Rejection is set to: 2
The number of strikes before interrupt is triggerd: 1
Disturber.
Disturber.
Disturber.
Disturber.
Disturber.
Disturber.
Disturber.
Disturber.
Disturber.
Disturber.
Disturber.
Disturber.
Disturber.
Disturber.
Disturber.
Disturber.
Disturber.
Disturber.
Disturber.
Disturber.
Disturber.
Disturber.

BG7UHS, 2021-05-06:
I also encountered the same problem. I don't know if it is due to a wiring error or other errors.  I use arduino nano

TadyTheFish, 2023-08-10:
I know this is an old issue, but if using the default examples, in the setup function you have
`lightning.resetSettings();`
at the end, you need to comment out that, or it will revert the detector to default settings

TadyTheFish, 2023-08-10:
No problem. I hit this snag also.. No matter what setting I tried it seemed that they had no effect... Then I tested this with printing the settings of the chip in the loop() and theye were default.. Then I removed the resetSettings and it worked