sparkfun/SparkFun_AS3935_Lightning_Detector_Arduino_Library #18: Isn't ZERO a valid setting?

LouPaloma, 2021-03-14:
Subject of the issue

Looking at the datasheets, shouldn't the valid settings include ZERO, e.g., noisefloor?

Steps to reproduce
`setNoiseLevel(0);`

Expected behavior
Valid setting

Actual behavior
Argument ignored.

```
void SparkFun_AS3935::setNoiseLevel( uint8_t _floor )
{
  if( (_floor < 1) || (_floor > 7) )
    return; 
  
  _writeRegister(THRESHOLD, NOISE_FLOOR_MASK, _floor, 4); 
}
```

shouldn't this be?:

```
void SparkFun_AS3935::setNoiseLevel( uint8_t _floor )
{
  if( (_floor < 0) || (_floor > 7) )
    return; 
  
  _writeRegister(THRESHOLD, NOISE_FLOOR_MASK, _floor, 4); 
}
```

Quoting datasheet:

> Table 16. Settings for the Noise Floor Threshold
> Continuous Input Noise Level
> [μVrms] (outdoor)
> Continuous Input Noise Level
> [μVrms] (indoor) REG0x01[6] REG0x01[5] REG0x01[4]
> 390 28 0 0 0
> 630 45 0 0 1
> 860 62 0 1 0
> 1100 78 0 1 1
> 1140 95 1 0 0
> 1570 112 1 0 1
> 1800 130 1 1 0
> 2000 146 1 1 1

smdll, 2021-08-31:
@LouPaloma agreed with you, however since the value is "uint8_t", a _floor < 0 check is not necessary

edspark, 2023-08-10:
Yes you are correct - this has been fixed in release 1.4.5. Unnecessary checks were also removed.

closed 2023-08-10 by edspark