sparkfun/SparkFun_AS3935_Lightning_Detector_Arduino_Library #19: maskDisturber function didn't set bit

djsirius, 2023-09-15:
### Subject of the issue
The disturber masking bit is not set, no matter if i set the value true or false. After i commented out the if statement lines in the file "SparkFun_AS3935.cpp".
https://github.com/sparkfun/SparkFun_AS3935_Lightning_Detector_Arduino_Library/blob/2dcc8bae6b1000c60191cee73bbeaefad5e9f213/src/SparkFun_AS3935.cpp#L263

// REG0x03, bit [5], manufacturere default: 0.
// This setting will change whether or not disturbers trigger the IRQ Pin.
void SparkFun_AS3935::maskDisturber(bool _state)
{
//    if ((_state == true) xor (_state == false))
//        return;

    _writeRegister(INT_MASK_ANT, DISTURB_MASK, _state, 5);
}

After that the bit got set correct.

### Your workbench
* What development board or microcontroller are you using?
ATSAMD21G18 based MCU

* How is the breakout board wired to your microcontroller?
AS3935 is connected via I2C.

* How is everything being powered?
doesn't matter

edspark, 2023-09-15:
I'll take a look ASAP, thanks.

edspark, 2023-09-15:
Sorry about that, made a logical misstep with the xor. I just went ahead and removed the check because the function can only take true or false as an argument. Fixed in the latest release.

closed 2023-09-15 by edspark