sparkfun/SparkFun_AS3935_Lightning_Detector_Arduino_Library #24: INDOOR OUTDOOR not work

elcharo, 2024-07-02:
I can´t write this option.  Return 0  in serial monitor.

// REG0x00, bits [5:1], manufacturer default: 10010 (INDOOR).
// This function changes toggles the chip's settings for Indoors and Outdoors.
void SparkFun_AS3935::setIndoorOutdoor(uint8_t _setting)
{
    if (((_setting != INDOOR) && (_setting != OUTDOOR)))
        return;

    _writeRegister(AFE_GAIN, GAIN_MASK, _setting, 1);
}

edspark, 2024-07-08:
Can you verify that you're using the correct values (listed below)? I just recently tested this particular function and it was working just fine. 

``` c
#define INDOOR 0x12
#define OUTDOOR 0xE
```

edspark, 2024-07-31:
Closing, but please reopen if you're still having problems

closed 2024-07-31 by edspark